FabianRig/ArduinoMoonBoardLED #1: LED Mapping

domenko, 2021-08-10:
Awesome! This looks really super promising for me. I was searching for an arduino solution before, now here it is! :)

But will it work also if I use a string of LEDs that skip every second LED (because of spacing?)

So:  ledmapping[] = {0, 2, 4, 6, 8, 10, ... ?

Thanks!

FabianRig, 2021-08-10:
Glad to hear! :-)

What I have tried is playing around with different LED mappings. And skipping every second LED does work on the software side. 

You would need to make to adjustments in the code:
`const uint16_t PixelCount = 400;`
and 
`int ledmapping[] = {0, 2, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30, 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 62, 64, 66, 68, 70, 72, 74, 76, 78, 80, 82, 84, 86, 88, 90, 92, 94, 96, 98, 100, 102, 104, 106, 108, 110, 112, 114, 116, 118, 120, 122, 124, 126, 128, 130, 132, 134, 136, 138, 140, 142, 144, 146, 148, 150, 152, 154, 156, 158, 160, 162, 164, 166, 168, 170, 172, 174, 176, 178, 180, 182, 184, 186, 188, 190, 192, 194, 196, 198, 200, 202, 204, 206, 208, 210, 212, 214, 216, 218, 220, 222, 224, 226, 228, 230, 232, 234, 236, 238, 240, 242, 244, 246, 248, 250, 252, 254, 256, 258, 260, 262, 264, 266, 268, 270, 272, 274, 276, 278, 280, 282, 284, 286, 288, 290, 292, 294, 296, 298, 300, 302, 304, 306, 308, 310, 312, 314, 316, 318, 320, 322, 324, 326, 328, 330, 332, 334, 336, 338, 340, 342, 344, 346, 348, 350, 352, 354, 356, 358, 360, 362, 364, 366, 368, 370, 372, 374, 376, 378, 380, 382, 384, 386, 388, 390, 392, 394, 396, 398};`

What I have *not* tried is connecting 400 LEDs to the Arduino (I only have 200 ;-)). 

In the FAQ of NeoPixelBus you can find some info (https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%237). My guess is that it should work because the Nano 33 BLE has quite a lot of RAM (256kB)...

FabianRig, 2021-08-11:
Sure, no problem! I'm excited to hear from you!

Actually, you can try and compile the project already. There's no need for the Nano if all you want to do is adapt the code and compile.

domenko, 2021-08-13:
I indeed tried that. It seemed to work. Not sure I would have been able to tell if it didn't :) 

May I ask how you wired the LEDs? Data and GND wires to the Arduino and separate power supplies for each? Or is there any danger of frying anything? :) Since I have a lot of LEDs to power I have a mean well mar 60-5 to provide 5V to the LEDs but the nano is a 3.3V so I'll power it separately. Or am I missing something? Cheers!

FabianRig, 2021-08-13:
Normally you should be able to tell if something went wrong because there are error messages in red letters. ;-)

Yes, I should have described that. Actually it's very easy with the Arduino Nano 33 BLE. While its operating voltage is 3.3 V, it can be powered with 4.5 V up to 21 V (see https://store.arduino.cc/arduino-nano-33-ble "tech specs"). Just power the LEDs and the Arduino with the same power supply at 5 V. The Arduino has a GND and VIN pin for that. Then connect the Arduino to the DATA line using a resistor (optional, I used 330 ohms). That's it.

(Technically, you might need a level shifter because the Arduino's logic level is 3.3 V and the LEDs level is 5 V. However, it just works for most people including me. ;-))

domenko, 2021-08-13:
VSC does show me 5 problems after compiling:

variable "uint8_t" is not a type name
and essentially four of these: no instance of constructor "RgbColor::RgbColor" matches the argument list -- argument types are: (int, int, int)

If I flash the Arduino it acts as a Moonboard (yeey!), and also translates the hold strings to corresponding LED map. Now all I need to figure out is why the lights don't turn on ... Perhaps it is something to do with the 5 problem messages? 

And I bet this is tedious to you, but would it be possible to see a picture of your wiring of the controller and the lights? I'd very much appreciate it! 

Cheers, d

FabianRig, 2021-08-13:
OK, that sounds like a problem. :-D
I do get a few warnings but they're all related to ArduinoBLE. The build is successful, however. Could you please post the entire file? I guess there's something messed up with the definition of the colors in the beginning.

As for the wiring: I'll attach a photo where you can see the connections (at least more or less ;-). GND of the power supply is connected to both the LEDs and the Arduino. 5 V is connected to both the LEDs and the Arduino. Pin D2 of the Arduino is connected (via 330 ohms) to the LEDs. And that's it. 

An no, it's not tedious. I'm very much looking forward to the first "stranger" trying the project! ;-)

[image]
[image]

domenko, 2021-08-16:
Brilliant! 

I've been tinkering back and forth (stubborn as I am) and I found that the problems were related to the referenced locations of my libraries (on OS X). After trying to compile and flash the Nano from a PC it worked first go, without issues (after I installed all the required libraries, etc.) 

And after that I wired up my test LED strand to the Arduino and it lit up! Everything seems to be working perfectly. As soon as my final LEDs arrive and are built into the Moonboard I will be sure to update you! :) 

Thanks a bunch for this project, good sir!

FabianRig, 2021-08-16:
Nice, I'm glad to hear that! 

You're very welcome. Please keep me updated whether everything works as expected!

smai86, 2021-08-17:
@FabianRig 
looks awesome!
I got https://github.com/lidiapb/moonboard this running. And works also pretty well. But now some friends asked me to build the same system for them. I think your project is easier to build up, because i worked a lot of hours on the other one to get bt running. ^^
So could u tell me where have u bought the LED Stripes? Here in Germany I only find the 7,5cm version and not 25cm... I think i would be easier to build the new systems with the 25cm led stripes, because I dont have to custimize your system.

Thanks a lot for your work on this!