FabianRig/ArduinoMoonBoardLED #3: Powering LED

Yetti20, 2022-11-09:
Great stuff here. I plan to play around and try to get it working on my home wall. For powering the LED do they get powered right through the Arduino. And can I just power the Arduino through the USB that connects it the laptop by plugging it into the wall? A bit of a newbie with some experience but very little.

FabianRig, 2022-11-09:
Thanks! Please have a look at the pictures where you should see the wiring.

The Arduino is not capable of powering all the LEDs! You definitely do need another power brick which provides power for both the LEDs and the Arduino at the same time. 

The Arduino works as a standalone device -- no connection to a laptop needed (once its flashed).

davidstyers, 2022-11-17:
@FabianRig First of all, thank you for this project! Due to the unavailability of the Pi Zero, I will be using this project to light my Moonboard.

Expanding on power supplies for the project, I'd like to hear about your experience using the LPV-60-5. Most other projects use the MDR-60-5 but at 5V, neither of which produce 60W (LPV-60-5 @ 40W, MDR-60-5 @ 50W). Did you choose this power supply because you are only driving 150 LEDs (Moonboard Mini)? Also, are you only noticing any discoloration due to voltage drop or are you using power injection?

FabianRig, 2022-11-18:
@davidstyers I created a separate issue so that the threads don't get mixed up. :-)

jordigrau83, 2022-11-22:
I have the standard 198 LED version + Arduino working with a basic 5v 1A
USB charger from an old phone, so yes you can use the VIN pin to power all
the LED strip

FabianRig, 2023-02-16:
Thanks for the info. A 5V 1A USB charger can work but I wouldn't recommend it. 

Especially if you power many LEDs at once or if you set the brightness to higher levels, it might not be enough...

closed 2023-02-16 by FabianRig