FabianRig/ArduinoMoonBoardLED #4: Power supply

FabianRig, 2022-11-18:
@FabianRig First of all, thank you for this project! Due to the unavailability of the Pi Zero, I will be using this project to light my Moonboard.

Expanding on power supplies for the project, I'd like to hear about your experience using the LPV-60-5. Most other projects use the MDR-60-5 but at 5V, neither of which produce 60W (LPV-60-5 @ 40W, MDR-60-5 @ 50W). Did you choose this power supply because you are only driving 150 LEDs (Moonboard Mini)? Also, are you only noticing any discoloration due to voltage drop or are you using power injection?

_Originally posted by @davidstyers in https://github.com/FabianRig/ArduinoMoonBoardLED/issues/3#issuecomment-1319055053_

FabianRig, 2022-11-18:
@davidstyers Glad to hear that you're interested in the project. I'm looking forward to more people using it!

Most probably you'll get away with any 5V power supply which ist able to drive the LEDs. Either by being powerful enough or by only powering fewer LEDs! (Or by lowering the brightness.) If your power supply were too small, you could just disable the LED check. Most Moonboard problems only have about a dozen LEDs and more than 20 holds seems to be very rare... I just chose a power supply which was available and not too expensive. On the Moonboard Mini, my power supply is absolutely sufficient.

As for voltage drop: I'm powering both sides of the string and I could inject power in multiple places. If you're using thick cables color fading shouldn't be much of a problem.

FabianRig, 2023-02-16:
Closing this.

closed 2023-02-16 by FabianRig

jordigrau83, 2023-02-16:
> (comment)

I drive the full standard Moonboard with a simple 5v 1A usb charger, and no
color loss by voltage sag, so you don't need anything else, its cheap and
easy to use

FabianRig, 2023-02-17:
Cool, very good to know! Thanks for the update.