FabianRig/ArduinoMoonBoardLED #5: Other System Boards

calvincheng919, 2023-01-31:
Hi Fabian, 

First of all, wonderful project. I linked to this project from the raspi version by e-sr https://github.com/e-sr/moonboard

This seemed way less complex.

I did have a question about how difficult it would be to port this to any of the other system boards out there (Kilter, Decoy, Tension, etc). I'm very new to BLE so unsure the ins and outs, but know that I likely will need to setup a packet sniffer to get log the data and somehow decipher it. Hoping they use similar/same backend as the Moonboard. 

Either way, I'll be building this as well... and if able to adapt the other system boards, it would be cool to have the arduino recognize which system board app is connecting and parse the incoming data. Not sure if I'm oversimplifying... but that's what I'm thinking. 

thanks! looking forward to hearing back from you.

FabianRig, 2023-01-31:
Hi Calvin!

Thanks. I'm glad to hear that.

From what I see, it all depends on the apps and protocol of the other boards. What this project does is "emulate" the BLE box, receive strings from the app (which thinks it talks to the real thing), decode them and turn on the corresponding LEDs. That's basically it...
 
What you would need to do: Find out, how exactly the other apps talk to the other boards' LED systems. My guess would be that the different apps won't even connect if the name doesn't match what they expect to see... You could visit a gym with another board and have a look at the list of BLE devices, perhaps you can find out more... ;-)

As for the protocol/decoding part: Once you receive BLE messages from the app, you need to find out which encoding is used and map that to the LEDs. That should be mostly reusable from this project!

Best regards,

Fabian

calvincheng919, 2023-01-31:
So from the very little I can decipher reading the HardwareBLESerial library code, when you pass in "Moonboard A" as an argument to beginAndSetupBLE method, it calls advertise()... So App comes along and "sees" that the led box is advertising "Moonboard A" and connects. 

Theoretically, if I find out Decoy Board's advertising name is "Decoy xyz" I should be able to advertise that and have the Decoy App connect and talk. 

Anyway, looks like I'm gonna be setting up my mac as a sniffer and visiting some gyms in the area 🤣

FabianRig, 2023-02-16:
I'm closing this. Feel free to reopen if the need arises!

closed 2023-02-16 by FabianRig