FabianRig/ArduinoMoonBoardLED #7: Circuit Diagram

aclel, 2023-02-09:
Thanks for this awesome project. I'm in the process of putting it together and I was wondering if you would mind sharing a circuit diagram? I'm interested in seeing exactly how you wired the power to the LED strings. Cheers!

calvincheng919, 2023-02-10:
I second that!

FabianRig, 2023-02-16:
Hi!

Sorry for taking so long to reply. Actually, it's quite easy and there's not much too it. The color of the cables may vary, though. It depends on the manufacturer... So double check if your manufacturer has any info on that.

GND (ground, negative):
- usually black cables (e.g. DC adapter)
- however, for my LED string it is blue!

Positive (+5V):
- usually red cables (e.g. DC adapter)
- however, for my LED string it is brown!

Data is yellow/green for my LED string. (Which actually is very weird...)

What you need to do:
- You connect all GND cables / pins of the LED string (blue for me), the Arduino (pin named "GND") and the DC side of the power supply (black for me).
- You connect all +5V cables / pins of the LED string (brown for me), the Arduino (pin named "VIn") and the DC side of the power supply (red for me).
- You connect the data cable of the LED string (yellow/green for me) to the Arduino (pin named "D2"). If you want to be sure, include a 330 Ohms resistor.
- Then you can just power the DC adapter and all should work.

That's it. 

If you're willing to draw a schematic, I could check that. 

Best regards

aclel, 2023-03-31:
Just wanted to say thank you for this project. Wired it all up in two hours this morning following your instructions, worked first go. I didn't drill any LED holes for row 1 due to it not having any holds by design in the 2020 minimoon set. I skipped two LEDs each time I changed column when placing them which was super convenient as I needed extra length in the string to get over the 180mm wide boards that I'd used for my uprights.

You're a legend, thanks heaps.

FabianRig, 2023-03-31:
Thank you very much for your kind words! I'm very glad that the projects helps other people, too. :-)

Exactly. Row 1 is not needed for the Mini version. I skipped several LEDs, too, in order to make the wiring hassle free. That's why I included the mapping. Way easier than drilling big holes and fiddling the cables through. ;-)

If everything works for you, feel free to close the issue.

closed 2023-04-15 by aclel

scott-lin, 2024-01-08:
I'm an electrical newb, so would you mind checking this diagram since you offered above? I just called the power lines positive and negative for simplicity since it seems like they can have different labels depending on the product. Hope it's clear enough.

Another question, you mentioned putting the resistor between the Arduino and LED data -- does the wattage rating matter for the resistor, or what do you recommend / use in your system?

[image]

FabianRig, 2024-01-09:
I opened a new issue for you and have a look at it as soon as I have time!