FabianRig/ArduinoMoonBoardLED #8: Bluetooth issues

Jstar018, 2023-05-28:
Hi Fabian, 

Thank you for pulling together. I am new to alot of this so apologies if this is an ignorant question. I have got the board, connected it up. compiled and uploaded the content to it but the Bluetooth is not visible on the moonboard app. 

I have been playing around for a while looking at differnet options. When I go to 'build' there are a number of error messages with the bluetooth setup (below is the output). I assume that this is what is driving it.  Wondering if it is me or there were some changes in the referene libs that this uses. 

Any advice and guidance on what I need to do to address is appericated. 

Cheers, 

Processing nano33ble (platform: nordicnrf52; board: nano33ble; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing SPI
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nano33ble.html
PLATFORM: Nordic nRF52 (9.6.0) > Arduino Nano 33 BLE
HARDWARE: NRF52840 64MHz, 256KB RAM, 960KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduino-mbed @ 4.0.2 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- HardwareBLESerial @ 1.0.0
|-- ArduinoBLE @ 1.3.4
|-- NeoPixelBus @ 2.7.5
Building in release mode
Compiling .pio\build\nano33ble\src\main.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLEAdvertisingData.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLECharacteristic.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLEDescriptor.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLEDevice.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLEService.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLEStringCharacteristic.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\BLETypedCharacteristics.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\local\BLELocalAttribute.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\local\BLELocalCharacteristic.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\local\BLELocalDescriptor.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\local\BLELocalDevice.cpp.o
In file included from .pio\libdeps\nano33ble\ArduinoBLE\src/utility/GATT.h:26,
                 from .pio\libdeps\nano33ble\ArduinoBLE\src\local\BLELocalCharacteristic.cpp:25:
.pio\libdeps\nano33ble\ArduinoBLE\src/local/BLELocalCharacteristic.h: In constructor 'BLELocalCharacteristic::BLELocalCharacteristic(const char*, uint16_t, int, bool)':
.pio\libdeps\nano33ble\ArduinoBLE\src/local/BLELocalCharacteristic.h:90:12: warning: 'BLELocalCharacteristic::_cccdValue' will be initialized after [-Wreorder]
   uint16_t _cccdValue;
            ^~~~~~~~~~
.pio\libdeps\nano33ble\ArduinoBLE\src/local/BLELocalCharacteristic.h:79:12: warning:   'uint8_t BLELocalCharacteristic::_permissions' [-Wreorder]
   uint8_t  _permissions;
            ^~~~~~~~~~~~
.pio\libdeps\nano33ble\ArduinoBLE\src\local\BLELocalCharacteristic.cpp:32:1: warning:   when initialized here [-Wreorder]
 BLELocalCharacteristic::BLELocalCharacteristic(const char* uuid, uint16_t permissions, int valueSize, bool fixedLength) :
 ^~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\local\BLELocalService.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\remote\BLERemoteAttribute.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\remote\BLERemoteCharacteristic.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\remote\BLERemoteDescriptor.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\remote\BLERemoteDevice.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\remote\BLERemoteService.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\ATT.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\BLEUuid.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\GAP.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\GATT.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\HCI.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\HCICordioTransport.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\HCIUartTransport.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\HCIVirtualTransport.cpp.o
Compiling .pio\build\nano33ble\lib6ac\ArduinoBLE\utility\L2CAPSignaling.cpp.o
[... 230 lines cut ...]
Compiling .pio\build\nano33ble\FrameworkArduino\api\Stream.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\api\String.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\arm_hal_random.c.o
Compiling .pio\build\nano33ble\FrameworkArduino\as_mbed_library\variant.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\itoa.c.o
Compiling .pio\build\nano33ble\FrameworkArduino\main.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\mbed\platform\cxxsupport\mstd_mutex.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\pinToIndex.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\random_seed.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\timer.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\wiring.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\wiring_analog.cpp.o
Archiving .pio\build\nano33ble\libFrameworkArduinoVariant.a
Compiling .pio\build\nano33ble\FrameworkArduino\wiring_digital.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\wiring_pulse.cpp.o
Compiling .pio\build\nano33ble\FrameworkArduino\wiring_shift.cpp.o
Archiving .pio\build\nano33ble\libFrameworkArduino.a
Linking .pio\build\nano33ble\firmware.elf
Checking size .pio\build\nano33ble\firmware.elf
Building .pio\build\nano33ble\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  26.7% (used 69920 bytes from 262144 bytes)
Flash: [===       ]  33.3% (used 327476 bytes from 983040 bytes)
======================================================= [SUCCESS] Took 45.36 seconds =======================================================
 *  Terminal will be reused by tasks, press any key to close it.

FabianRig, 2023-05-30:
Those are "only" warnings and in the end it says "SUCCESS". It might still work. 

Could you connect the Nano to your computer and check the Nano's console logs? What does it say?

FabianRig, 2023-06-05:
Sorry for taking so long to answer. Currently, I don't have a lot of time... Sorry!

Do you have an original "Arduino Nano 33 BLE" or another brand?

Have you tried starting totally fresh? Reinstalling VS Code, Platform IO and all packages?

FabianRig, 2023-06-05:
I just gave it a try in a new environment. There are indeed 45 warnings with respect to the libraries used. Warnings aren't errors however and the build is successful:

RAM:   [===       ]  26.7% (used 69920 bytes from 262144 bytes)
Flash: [===       ]  33.3% (used 327476 bytes from 983040 bytes)
=========================== [SUCCESS] Took 38.53 seconds ===========================

The problem you're seeing might be something else. As soon as you successfully flash that file, you should see messages in the terminal.

Jstar018, 2023-06-14:
Hi Fabian, 

Apologies, I have been traveling. Yes I am using an original "Arduino Nano 33 BLE"

I have reinstalled VS Code, Platform IO and the project. 

Interestingly I am getting 47 error messages not the 45 you got on your build.  Build size etc is exactly the same. 

Still crashing and not able to monitor the COM Port or link in with Blue tooth....

Jstar018, 2023-06-15:
I can flash without errors 
I can connect to the right com port with only the USB connection and nothing else connected to the PC
I can flash another project and monitor it 

I have not tried with another PC...

FabianRig, 2023-06-15:
OK. That's a good start. If you can successfully flash it without errors and connect to it without any error, then it should work. (Otherwise you should see "BLE setup failed!" -- check https://github.com/FabianRig/ArduinoMoonBoardLED/blob/main/src/main.cpp line 29.)

What puzzles me is that the MoonBoard app doesn't see the device. I've used many different Android devices (mobile phones, tablets) and friends are successfully using various iPhones. The range isn't too good (less than 4 or 5 meters for me), other than that it's stable and reliable.

If you're using an Android phone: Please make sure the app has the location permission which is needed for bluetooth. Perhaps try reinstalling the app? Or could you maybe try another mobile device?

FabianRig, 2023-06-15:
One more idea: If you're using an Android device, please download nRF Connect for Mobile (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=de&gl=US) and use the scanner. You should see "MoonBoard A". This app also needs the location permission.

sviga, 2023-06-18:
I also had a problem with bluetooth, neither moonboard app nor nRF Connect for Mobile was discovered the device. But the issue was with **nordicnrf52** v.9.6.0 platform (see issue [here](https://github.com/platformio/platform-nordicnrf52/issues/175)), so I downgraded it to 9.5.0 and it started to work.

FabianRig, 2023-06-18:
Thank you for letting us know! Great find!

@Jstar018 Could you please change the line to "platform = nordicnrf52@9.5.0" in platformio.ini (https://github.com/FabianRig/ArduinoMoonBoardLED/blob/main/platformio.ini) and check if it works? If yes, we have a fix.

Jstar018, 2023-06-18:
Hi @FabianRig @sviga 

I downgraded the platform and it works perfectly. I can see the output in the monitor with the correct LEDs referenced. Bluetooth connects. 

Thank you!

jordigrau83, 2023-06-19:
I have used this app on many different arduino BLE 33 without any issue,
but if you want you can buy a cheaper ESP32 on AliExpress, and it also
works flawlessly, so if you still have problems with the Arduino try the
ESP32 😉

FabianRig, 2023-06-20:
Great! Good to know that it works now. I'll include the fix asap.

And thanks for the info regarding the ESP32!

closed 2023-06-20 by FabianRig

lbebenek, 2025-04-28:
Welcome.

Sorry for this but i have original board adafruit father 32u4 but erased software . I locking orinal software to download to this board. Can someone have soft. From original board please share . 

Best.

lbebenek, 2025-04-29:
I buy nano 33 and put your git project. I hope work with original Moon leds. I look orinal clone software for orinal board Adafruit. Because unfortunatly erased .  At this moment price for new "control box" 600 eur is too much.  I have one request can you implement reset button like orignal moon for reset connections on ble? Or in your opinion reset board is ok for users ? Best!!!

FabianRig, 2025-04-29:
Please open a new issue for further questions. I'll probably not add a reset button. You can simply power cycle it.