FabianRig/ArduinoMoonBoardLED #9: Fun...

Jstar018, 2023-07-02:
Hi Fabian, 

Board it working perfectly! Love the setup. 

This is not an issue with any of the functionality for the moonboard but could not message directly. 

I am wondering if there are some simple commands I can use via LightBlue or BlueFruit (or another app) that can be used to control the LEDs e.g.  turn on all the LEDs to xxx colour, have them rotate like a rainbow. I have two kids and thought doing some things like this will get them super excited about spending more time out there with me climbing. 

Thanks, I know this is a left field question.

FabianRig, 2023-07-03:
Hi! 

Unfortunately, this is not possible with this project. The software decodes strings ("text") which is sent by the MoonBoard app to light up the problem sequence. Therefore, it doesn't work with other apps which control the LEDs in another way. Sorry!

Best regards,

Fabian

closed 2023-07-03 by FabianRig

jordigrau83, 2023-07-03:
Hi, i added buttons on mine to do some extra tasks like switch from last
loaded projects or iluminate upper and lower LED of each hold so you can
see it from under or over it when climbing, i also added light animations
to make it more fun

Jstar018, 2023-07-03:
@jordigrau83 - can you share what you did? 

I was going to move some of my leds as it is hard to see from above. Lighting both is a good idea.

FabianRig, 2023-07-04:
If you want to light both the bottom and the top LED that's already supported out of the box. 

Just check "Do you have a V2 MoonBoard?" and enable the option for lighting both LEDs (above and below). It's already included in the project. ;-)

I'm not planning on adding additional complexity to the project (e.g. buttons, light shows). This project should remain as simple as possible. But please feel free to clone it and add any features you want. :-)