FabianRig/ArduinoMoonBoardLED #11: Has anyone tried with the official led lights + Arduino combo?

FabianRig, 2023-10-07:
Has anyone tried with the official led lights + Arduino combo?

_Originally posted by @marcosncosta1 in https://github.com/FabianRig/ArduinoMoonBoardLED/issues/10#issuecomment-1751623821_

FabianRig, 2023-10-07:
I haven't tried (don't have the original LEDs) and to be honest, I don't see why one would like to do this. If the original Moonboard system works, this projects doesn't offer any benefits. If you're planning on buying, the all-in-one package is only sold by Moonboard, this project might be cheaper but requires some work.

marcosncosta1, 2023-10-11:
I agree. I just have not found the LEDs except from China and with the shipping it ends up the same price. This project still makes the system much more affordable even with the official LEDs. The controller and power supply alone cost $600.

jordigrau83, 2023-10-13:
Where are you from? I foud a supplier from china that made me the led
strips with 26cm between leds. If you are from Europe i can send you a set

teamwork911, 2023-10-21:
https://www.aliexpress.us/item/3256801674884092.html?gatewayAdapt=glo2usa4itemAdapt

I just built mine, the LEDs take about 2-3 weeks from china but its only about $75 for a mini/ around $100 for a full size

marcosncosta1, 2023-10-22:
@jordigrau83 I am currently based between Switzerland and Germany. I actually ended up ordering the official lights, but if I can get my hands on something that works just as well I could order a refund since they didn't even process the order yet. I have a strong intuition they also source their lights in China and basically drop ship them. That's why there's a lead time of 2-3 weeks from them as well.

jordigrau83, 2023-10-23:
Ok, thanks for the reply, i ordered more leds from china and i'll have them
soon, so if you want i can send you a set for 100€, I'm installing them in
a different way than standard, its slower to install but the end result is
way better, i can send you pictures of a finished moonboard if you want

marcosncosta1, 2023-10-25:
Hey @jordigrau83 can you DM me on whatsapp +34695117657

marcosncosta1, 2023-11-04:
I have an update. I got he Moonboard LEDs and they are exactly like the others you can get on Amazon or Aliexpress(I also bought a 25 pixel LED to test beforehand). 
After setting up the lights and connecting the arduino I realized it does not work properly. I have the 1st light on as red and all of the rest blue. 
Anyone experienced this before?
I am running a full moonboard on the new updated app.

[image]
[image]
[image]

marcosncosta1, 2023-11-05:
As a side note, when the arduino is not connected all of the lights are blue.

marcosncosta1, 2023-11-05:
new update. It works perfectly with my test light now, but not with the moonboard lights yet.

https://github.com/FabianRig/ArduinoMoonBoardLED/assets/34763646/a3dd6a69-3700-49bc-9a5f-4312a16df8de

FabianRig, 2023-11-05:
Your videos looks good. That's what it's supposed to look like. 

A few ideas:
- From your picture I couldn't see whether GND is properly connected. Please check the pictures in the README: I power both the LEDs and the Arduino by one power source and thus, GND is properly connected. I do not use the USB connection except for flashing and debugging.
- The original/official MoonBoard LEDs might actually use a different controller! Have a look at https://github.com/Makuna/NeoPixelBus/wiki#supported-pixels and check if you can find out which controller the original LEDs use.

marcosncosta1, 2023-11-05:
Yes, I am powering through the usb. Thanks for the heads up on the controller. I'll share a video later of the lights. I did manage to get a response but it seems the light mapping is completely different.

marcosncosta1, 2023-11-05:
I just realized I didn't even bother looking at the specs of the LEDs since they looked the same however they are not WS2811 with F8 chips (which is what the test lights I got are) they are actually WS2811 IC F5 chips. So this might be the problem. I will investigate a bit further. 
[image]

marcosncosta1, 2023-11-05:
Soved it!! I powered the Nano as you suggested and that did the trick. I also read up on the library you shared and they specifically say that you have to wire the ground and data cables to the same source otherwise it will not work. 
Thanks for the tips and all the help. 
Now we know....The original lights do work with this system. But you might as well order the cheaper ones from Aliexpress....
Good climbing to you all!!!

FabianRig, 2023-11-05:
Cool, thanks for testing it! I've already updated the README so that it is clear that GND needs to be connected.

You're welcome. :-) I'm happy that it works for you.

Good climbing to you as well!

closed 2023-11-05 by FabianRig