FabianRig/ArduinoMoonBoardLED #12: New moonboard up doesn't light up green blue lights

sintoris, 2023-10-17:
I finished building my mini moonboard last week, just about when the new moonboard app was released. The old app works great with this project. With the new app, only the start (green/violet) and finish (red) holds are lighted up. The holds in between remain dark. 

Any clues as to what could be wrong?

PS: Thanks for this great project!

sintoris, 2023-10-17:
This was easy to fix. Just changed "P" to "R" in main.cpp:157 and :198.
 
In the new moonboard app the progress holds are marked with "R". In the old app it was with "P".

For example, for the problem "70-250!" the problem strings are:
Old app Problem string: S74,S114,P39,P103,E11,E60
New app Problem string: S74,S114,R39,R103,E11,E60

FabianRig, 2023-10-18:
Thanks for letting us know! I assume you're using an iPhone? I don't remember seeing an update for my Android device.

I'm on vacation at the moment and don't have access to my laptop. Therefore I can't fix it right now.

However, I'd suggest we change both lines to an OR statement and check for both P and R. Like that, the project works with any version.

sintoris, 2023-10-19:
No I am using Android. The old app has been silently removed from the play store last week. A new app has appeared with slightly different icon. One can actually have them both installed at the side by side but only if you have had installed the old app before last week. Confusingly the version numbering has restarted from 1.x with the new app. The old one stopped at 8.something.

I think there is no need to keep both P and R in the code. First, one cannot install the old app anymore. Second, even if you have the old app installed, you can't log any sent problems with it anymore. At least that's the case with me.

marcosncosta1, 2023-10-22:
Nice catch. I am about to start on the project. I will make the changes in the code as suggested

OllieCMartin, 2023-10-23:
The new app introduces 'show move beta', @sintoris is right about just changing "P" to "R" during normal operation, but when show beta is on you get additional holds/colours:

"L" -> lefthand
"R" -> Righthand
"M" -> Match hands
"F" -> footholds

so you have S, R, L, M, F, E hold types now! 

Easy enough to fix, I'm not sure what the right colours should be so I ended up just making them up!

FabianRig, 2023-10-29:
Thank you all for bringing this up!

If I understand correctly, there are no more "progress" (P) holds but "left" (L) and "right" (R). If show beta is not enabled, the only type of progress holds are "right" holds, correct? (I'll be able to check after my vacation...)

If anybody knows what colors the original system shows, I'd be happy to implement that. Otherwise I'll suggest, we use blue for right and yellow for left.

FabianRig, 2023-10-29:
I'm closing this issue because it should be fixed. Feel free to open a new one if there are any issues.

closed 2023-10-29 by FabianRig