FabianRig/ArduinoMoonBoardLED #13: help

lucadona98, 2023-10-26:
please can you explain to me how to download the ArduinoMonnboardLED project on platformio.
and from that passage onwards

please, I'm going crazy

marcosncosta1, 2023-10-29:
The steps are described in the ReadMe file in detail. I can take some photos of the specific steps on the Visual Studio Code for you.

FabianRig, 2023-10-29:
Hi @lucadona98!

As @marcosncosta1 wrote: you can find all details for both Windows and Linux in the README. For downloading, you can use git (clone the repository) or simply download the repository as a zip file!

Please let us know if that worked for you.

lucadona98, 2023-10-30:
then I put the program on platformio. I start a run and it tells me that there are some packages missing to install, I install the missing packages, but it always tells me the same thing that the packages are missing.

FabianRig, 2023-10-31:
Please follow the instructions in the README. 

If there are any errors along the way, copy/paste them here (and format them as code). Explain in detail what you did (what buttons/menus you clicked, what happened etc.). We need to know the details in order to help you. "some packages mssing" is not specific enough.

FabianRig, 2023-12-17:
It seems that I can close this issue. Feel free to re-open it.

closed 2023-12-17 by FabianRig