FabianRig/ArduinoMoonBoardLED #14: Trouble Flashing

b00kw0rm451, 2023-11-23:
Hi, this is my second github project, my first was a magicmirror. I'm working on a Rapsberry pi 4b. I downloaded and installed Visual Studio Code, I installed PlatformIO and git cloned the project. We have a standard Moon board so I didn't adjust any config.h settings. 

When I plug in the arduino and open the project and click "upload" under project tasks I get the following error:
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.80201.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.80201.0' requirements for your system 'linux_aarch64'

All of this is new to me, I would love help figuring out this error. We have the powersupply and lights all ready to go. Thank you!

FabianRig, 2023-11-23:
Hi!

That error means that there is a dependency issue. The RPi uses an ARM processor and therefore, other  packages are used than for an x64 platform. 

Two things you could try: 
- Use another computer / laptop that doesn't have an ARM architecture. Then the build chain should work out of the box.
- Change "nordicnrf52@9.5.0" in platform.ini to "nordicnrf52" in order to remove version pinning. This is generally not recommended and can lead to other problems but might help in this case.

b00kw0rm451, 2023-11-24:
Hello,

Thanks so much, I added Visual Studio Code to my Mac, and successfully flashed! Now for the next challenge. I'm having the same problem as Sintoris, only the start and finish holds are lighting up. I pulled the code from github this morning and I know it says it was updated...not sure what I'm missing.
[image]

FabianRig, 2023-11-24:
Could you please check the following:
- Are you using the most recent app from your app store? There is NO update for the older app, instead, they removed it and replaced it with a new one.
- Connect the Arduino to your laptop and paste a log of the problem string here so that we can see if there are other hold types missing.

I assume that you're sending old problem string, per "P" were progress holds. Now we have "L", "R" and so on.

b00kw0rm451, 2023-11-24:
Wow...it really was the app. My partner has been using it at the local gym and I never asked which version it was. That's all we had to change. Thank you for creating this and for your help! It's his dream come true.
[image]

FabianRig, 2023-11-24:
I'm very happy to hear that. :-) Congratulations to your nice board!

Have fun climbing and don't get hurt. ;-)

I'll close this issue but feel free to open another if you run into any trouble.

closed 2023-11-24 by FabianRig

FabianRig, 2023-11-24:
I reintroduced the 'P' type of hold. It should work with the old app again -- however, as far as I know, the old app isn't available and usable anymore... So if your buddy wants to use his old app, it should still work.