FabianRig/ArduinoMoonBoardLED #18: How many leds?

tramas76, 2023-12-20:
Hi, reading the readme file you specified 150 LEDs for moon board mini and 200 for moon board.
Moon board mini should be 132(11x12) and 198(18*11) moon board.
Am I doing something wrong?
Thanks

Lorenzo

FabianRig, 2023-12-21:
Hi Lorenzo,

You're not wrong! What you're talking about are the number of LEDs visible or number of "LED holes" in the board. That is correct. However, the LED strings usually come in lengths of 50. So for the Mini you need three strings of 50 (150) and for the full size you need four strings (4 * 50 = 200). If you have more LEDs in your string, the rest will just be ignored and its usually cheaper to just by off the shelf LED strings.

Does that help?

Best regards!

tramas76, 2023-12-21:
Hi Fabian! Thank's for reply and congratulations for your job!

ok, but can I cut the LED strip to 132?..
..or should I keep the LEDs up to 150 anyway?

FabianRig, 2023-12-22:
Absolutely no need to cut it. Just leave it at 150 LEDs. :-) 

If you really want to cut it: I guess, it should work -- but there's no need to do it.

closed 2023-12-22 by FabianRig