FabianRig/ArduinoMoonBoardLED #19: Are multiple user connects at the same time supported?

scott-lin, 2024-01-08:
The official Moonboard LED system "Allows up to 7 users to connect at the same time" according to their product page: [MoonBoard LED System](https://moonclimbing.com/moonboard/led/moonboard-led-system-v3.html)

Your readme says all features are supported in the latest app as of Nov 2023. I wanted to ask whether this specific feature of supporting multiple users is possible, however, since your readme just mentioned "show beta" and "LEDs above and below" features.

If it's not supported currently, would a feature request of this kind be considered?

Thanks for the great work put into this project!

FabianRig, 2024-01-09:
Hi Scott,

To be honest: I don't know if it still works. Last time we checked, it worked at least for two devices if I remember correctly (Android and iPhone at the same time). You'll have to try yourself.

No, I have no plans on addings this feature. This project is meant for personal use, not for commercial gyms. For commercial gyms I strongly recommend the official solution.

Best regards

scott-lin, 2024-01-09:
Got it, thanks. I was more asking for a husband and his wife scenario at home as opposed to a gym. I'll test it out if I get it all up and running on my board.

closed 2024-01-09 by scott-lin

sintoris, 2024-01-11:
Hi, 

I have just implemented this feature last weekend, for my mini moonboard, based on this project, for the same reason. We have not had the chance to actually use it with two smartphones yet.

To do that actually need to have a second arduino nano ble. One can not be used with two names (e.g. "MoonBoard A", "MoonBoad B") at the same time. Just connect the output of the each of the two arduinos to the data wire of the output pins. Between each output pin and the data wire you need to have a schottky diode. 

I have named one arduino "MoonBoard White" and the other "MoonBoard Purple". The app doesn't seem to care what comes after the "MoonBoard" string. The mini moonboard uses not more than 110 holds. So the remaining LEDs can be lighted up either in white or purple color depending on which arduino sent the problem to the strip.

Not sure how to share this code if you want it. Although it is not too much work, I ended up with many changes in @FabianRig 's project, basically moving some code to functions to make it easier for me. Also, I haven't made the effort to prevent interference, e.g. if both arduino's send data to the strip at the same time. I started to, but debugging was a pain and I don't really think it will be much of a problem.

FabianRig, 2024-01-12:
Hi all,

Thanks for bringing this up again!

@sintoris: If I understand you correctly, you're just using two Arduinos which share the same data line and which can talk to the LEDs at the same time. You just added diodes to stop one Arduino from talking to the other, correct? If there is no collision handling, there wouldn't be any need to adjust the code. It would be a simple hardware solution to a problem where there is no easy software solution. ;-)

I had a look at the underlying libraries. HardwareBLESerial provides a Singleton, so that would be one thing to change. We could reimplement the serial connection using the underlying ArduinoBLE library. The key problem seems to be that the library stops advertising the service as soon as there is one connection. Technically, multiple connection should be possible.

I don't have too much time on my hands and I don't see a very easy way to do it. Therefore, I don't think I'll implement it soon. It is also a trade-off between simplicity and functionality. Right now, this project is extremely simple and works perfectly for a single user (or phone). Perhaps just adding hardware (one more Arduino) is indeed the simpler solution.

Best regards

sintoris, 2024-01-12:
@FabianRig exactly. I spent some time trying to figure out how to do it easily with just the one Arduino. Once I realized I was out of my depth, I just bought a second Arduino to do the job. 

Thanks again for the great project!

FabianRig, 2024-01-12:
Great, thanks for clarifying. I think it is a valid and pragmatic solution!

Do you have a part number for the diodes? I'd add that to the README, in case somebody absolutely needs multi user capability.

sintoris, 2024-01-12:
Sure, I used a pair of Schottky diodes 1A 1N5819.

FabianRig, 2024-01-14:
Thanks. I've added that to the README.

jordigrau83, 2024-01-15:
hi, the solution is way simpler than using 2 arduinos and diodes: the
problem with multiple connections is in the BLE library, not on the
MOONBOARD project. That library stops advertising the "Moonboard A"
bluetooth service when there is 1 connected device, so the solution is
advertising again, you have to make this simple change on the
HardwareBLESerial.cpp:
modify the begin() to add the ConnectHandler and the DisconnectHandler, and
add this 2 simple handlers to advertise again :)

void ConnectHandler(BLEDevice central) {
  // central connected event handler
  Serial.print("Connected event, central: ");
  Serial.println(central.address());
  BLE.advertise();
}

void DisconnectHandler(BLEDevice central) {
  // central disconnected event handler
  Serial.print("Disconnected event, central: ");
  Serial.println(central.address());
  BLE.advertise();
}

void HardwareBLESerial::begin() {
  BLE.setAdvertisedService(uartService);
  uartService.addCharacteristic(receiveCharacteristic);
  uartService.addCharacteristic(transmitCharacteristic);
  receiveCharacteristic.setEventHandler(BLEWritten, HardwareBLESerial::
onBLEWritten);
  BLE.addService(uartService);

  BLE.setEventHandler(BLEConnected, ConnectHandler); // *** per permetre
multiples connexions
  BLE.setEventHandler(BLEDisconnected, DisconnectHandler);
}

jordigrau83, 2024-01-15:
> The official Moonboard LED system "Allows up to 7 users to connect at the same time" according to their product page: [MoonBoard LED [...]

Hi, the problem with the original version is that the BLE library is not
advertising the ble server when a device is connected, so no multiple
connections are allowed, i managed to do it adding very simple code on the
BLE library, HardwareBLESerial.cpp:

void ConnectHandler(BLEDevice central) {
  // central connected event handler
  Serial.print("Connected event, central: ");
  Serial.println(central.address());
  BLE.advertise();
}

void DisconnectHandler(BLEDevice central) {
  // central disconnected event handler
  Serial.print("Disconnected event, central: ");
  Serial.println(central.address());
  BLE.advertise();
}

void HardwareBLESerial::begin() {
  BLE.setAdvertisedService(uartService);
  uartService.addCharacteristic(receiveCharacteristic);
  uartService.addCharacteristic(transmitCharacteristic);
  receiveCharacteristic.setEventHandler(BLEWritten, HardwareBLESerial::
onBLEWritten);
  BLE.addService(uartService);

// create this Handlers to advertise BLE server again:

  BLE.setEventHandler(BLEConnected, ConnectHandler);
  BLE.setEventHandler(BLEDisconnected, DisconnectHandler);
}

sintoris, 2024-01-16:
Hi, this is great! I 'll give it a try.

One question, does this solution advertise "MoonBoard A" to all centrals?

FabianRig, 2024-01-16:
That's a great solution. I'll have a look at it and will include it in the project.

It seems that there is no easy way to patch a library in Arduino. Therefore, I'll probably include the HardwareBLESerial files in this project and change the files locally. If you know a better way, please let me know.

I've created another issue to track the changes there: https://github.com/FabianRig/ArduinoMoonBoardLED/issues/21

FabianRig, 2024-01-18:
The "new" firmware seems to work fine. Advertisement continues as planned. Could you please check and report here: https://github.com/FabianRig/ArduinoMoonBoardLED/issues/21 whether it works as expected?