FabianRig/ArduinoMoonBoardLED #20: Wiring diagram

FabianRig, 2024-01-09:
@FabianRig 

I'm an electrical newb, so would you mind checking this diagram since you offered above? I just called the power lines positive and negative for simplicity since it seems like they can have different labels depending on the product. Hope it's clear enough.

Another question, you mentioned putting the resistor between the Arduino and LED data -- does the wattage rating matter for the resistor, or what do you recommend / use in your system?

[image]

_Originally posted by @scott-lin in https://github.com/FabianRig/ArduinoMoonBoardLED/issues/7#issuecomment-1881972656_

FabianRig, 2024-01-09:
@scott-lin:

I had a quick look at the wiring diagram. The wiring itself looks correct. However, I have a few remarks:
- the AC side doesn't have pos or neg but L and N, colors would be blue for N and brown for L (in Germany)
- the DC side should be +5V and gnd (ground)
- colors on the DC side should be black for gnd and red for +5V
- the LED strings usually have weird colors ;-)

You can use any kind of 330 Ohms resistor, no special resistors needed. There won't be any large currents -- otherwise the Arduino would be dead anyway.

scott-lin, 2024-01-09:
Thanks, updated per your comments and my region's AC colors.
 

[image]

DenkBrettl, 2024-01-09:
Thanks Scott! Would you be ok to add 230V as well? E.g. "120V or 230V power supply". 230V is the standard in a lot of the world and it might confuse some people if the diagram was very specific.

Thanks, Klaus

scott-lin, 2024-01-09:
> Thanks Scott! Would you be ok to add 230V as well? E.g. "120V or 230V power supply". 230V is the standard in a lot of the world and it might confuse some people if the diagram was very specific.
> 
> Thanks, Klaus

Sure, I hope this update covers the gambit.

[image]

DenkBrettl, 2024-01-09:
Looks great - thanks! :-)

FabianRig, 2024-01-10:
If it's OK for you, I can include the diagram in the readme?

scott-lin, 2024-01-10:
Yes, please feel free.

FabianRig, 2024-01-12:
Thanks for creating and providing the wiring diagram. I've added it to the README. :-)

tmotyl, 2024-01-15:
Thanks for the diagram. I've read somewhere that you have to connect power  every x number of leds? is that right? if yes it would be awesome to have it on the diagram too

FabianRig, 2024-01-15:
No, it's definitely not necessary for the Moonboard LEDs. If you want to light up all LEDs, there might be a difference in color/brightness. For the handful of LEDs it's not necessary.

If you absolutely want to do double wiring, you can just connect both ends of the LED string to GND and +5V.