FabianRig/ArduinoMoonBoardLED #21: Multi user support

FabianRig, 2024-01-16:
TODO: Include changes to library so that multi user support is possible. 

See: https://github.com/FabianRig/ArduinoMoonBoardLED/issues/19#issuecomment-1892834666

FabianRig, 2024-01-16:
I've included a fork of the original BLE serial library in https://github.com/FabianRig/ArduinoMoonBoardLED/commit/07d139b793f74d3f4cbf728fd50fa7d362962ec6. That should hopefully enable multi user support. 

If further testing shows that everything is working correctly, I'll update the README accordingly.

Aiasa21, 2024-02-02:
Hi FabianRig, I'm using your code sussefully on my home moonboard. I don't understand where I can find the last code that implement the multi user support.
Thank you for the help!

FabianRig, 2024-02-03:
There are no code changes in the project but in the dependencies. Have a look at https://github.com/FabianRig/ArduinoMoonBoardLED/blob/main/platformio.ini where you can see that one dependency has been replaced with a custom version (forked and patched by me).

Just make sure you have the most recent version of this repo (git pull or git clone to a new directory) and that's it.

FabianRig, 2024-02-03:
Good to hear that it's working for you. But that's *not* the intended way. You can simply clone this repo -- no manual adjustments needed!

Aiasa21, 2024-02-03:
I'm sorry about that, actually I don't know how to do so yesterday I prepared the edit manually.
As soon as I have some minutes I'll try to figure out how to do it and update directly in the visual studio project.

scott-lin, 2024-02-03:
Should the hardware solution for multi-user support be removed now that you have added a software solution? Namely, the readme updates from commit https://github.com/FabianRig/ArduinoMoonBoardLED/commit/3fd616aa6f5737524b4cbe91d53dd951f88b4353.

FabianRig, 2024-02-03:
> I'm sorry about that, actually I don't know how to do so yesterday I prepared the edit manually. As soon as I have some minutes I'll try to figure out how to do it and update directly in the visual studio project so to give you a feedback.

No need to apologize. :-) It's just more work to manually change code when you can just git clone / git pull. If you want a simple solution: just download this repo again to another folder and build and flash the firmware again.

FabianRig, 2024-02-03:
> Should the hardware solution for multi-user support be removed now that you have added a software solution? Namely, the readme updates from commit [3fd616a](https://github.com/FabianRig/ArduinoMoonBoardLED/commit/3fd616aa6f5737524b4cbe91d53dd951f88b4353).

You're absolutely right. I changed the README.

closed 2024-02-03 by FabianRig

scott-lin, 2024-02-11:
Just wanted to drop by again and say thanks for this particular update! It works great, and my wife and I are enjoying it.

FabianRig, 2024-02-13:
Thank you for your kind words! :-) I'm glad it's working for you!

Aiasa21, 2024-02-13:
> No need to apologize. :-) It's just more work to manually change code when you can just git clone / git pull. If you want a simple solution: just download this repo again to another folder and build [...]

I did this and I confirm that it work good, great job.. Yesterday we climbed with 4 app simultaneously, amazing! 

Just to know, there's a maximum connection limit?

FabianRig, 2024-02-13:
Nice, that's good to know! To be honest: I assume that there is a limit, but I don't know what the limit is. There is a limit for concurrent *connections* in the underlying libraries. However, the BLE Serial is not really a "connection" in that sense, so I'm not sure whether that internal limit applies. 

Four parallel connections is already impressive. :-D We'll have to round up as many devices as possible and try it out. ;-)

Aiasa21, 2024-02-13:
I understand, maybe I could try with 5 connection together if one day we'll climb together, I'll you know if this will happen. Anyway I suggest to disable the setting "Light problem on swipe" otherwise is a mess :D

If would like to suggest one update: 
it would be possible in the future to set a lower brightness during the start cycling so as to avoid an excessive use of current and then turn the brightness high during the climb?
That's because the section of the wire used by the led is extremely low and while all the leds are on the current is really high. 

Thanks again for your work!