FabianRig/ArduinoMoonBoardLED #23: Compiling issue

DashCodingSecondAcc, 2024-05-16:
Hi. 
i am following the Linux install ReadMe and everything works fine, except for the compile step, which gives me this error:
sh: 1: arm-none-eabi-g++: not found

[image]

please help. Great project, would love to get itworking!

DenkBrettl, 2024-05-16:
At a first glance I suspect it has to do with you using a RaspberryPi. Any chance to try on an x86 machine? If not, could you search the internet for the error and try to follow the advice you can find for it?

DashCodingSecondAcc, 2024-05-22:
Fixed it :) it was my bad - I just used VS code instead.

closed 2024-05-22 by DashCodingSecondAcc